Adarsh2425/IMEX-Independent-Multi-Extrusion-3D-printer- #7: the 2 heads works on the same piece or on 2 different pieces ?

k-lip12, 2023-09-04:


k-lip12, 2023-09-04:
same piece is very case specific, multiply seams and accumulate problems (ultra precise stepper, since they need to be in perfect synchronicity). The current hardware is far from being compliant with this kind of technical requirement.

-> Different piece is just emulating 2 printers, but you do that by dividing print surface by about 2. It saves space, but I guess this is not the target here.

Adarsh2425, 2023-09-04:
it will be possible this printer both the head will be on a single z axis support and i was thing if i am already thinking of having a custom processing unit i will have ultra sound sensor to show bed leveling error or also using some commands to adjust it.