Adarsh2425/IMEX-Independent-Multi-Extrusion-3D-printer- #9: this configuration both tools have a common Z, so unless layer times are more or less equal you're gonna waste a lot of time waiting for the other tool to finish and introduce oozing problems and have to constantly heat and cool nozzles.

k-lip12, 2023-09-04:
Also you mention wanting 2 tools to print different things simultaneously, which is indeed a problem for IDEX machines. However in this configuration both tools have a common Z, so unless layer times are more or less equal you're gonna waste a lot of time waiting for the other tool to finish and introduce oozing problems and have to constantly heat and cool nozzles. You also need to run the same material in both tools because bed temp would have to be identical.

If you want to print 2 different objects, using 2 printers just makes way more sense, and for multi material build a toolhead switcher/IDEX (IDEX if you print batches of parts)

k-lip12, 2023-09-04:
that's y I am creating custom software to minizine the wait time or nullify it by just assigning both the heads same amount of work so that it doesn't have to wait. solving oozing problems.

Adarsh2425, 2023-09-04:
that's y I am creating custom software to minizine the wait time or nullify it by just assigning both the heads same amount of work so that it doesn't have to wait. solving oozing problems.